Xinyuan-LilyGO/LilyGo-T-OI-PLUS #5: How to measure battery level in code?

gigamegawatts, 2021-08-15:
Is the battery connected to an ADC on the ESP32-C3, so we can measure its level?

gigamegawatts, 2021-08-17:
Now that they've added the schematic, I can answer my own question.  The battery is connected to A0 (ADC1_CH2) through a 1:1 voltage divider.  

I've confirmed that the voltage value on A0 is correct using a multimeter.  (I haven't read up on the C3's ADC yet, but the ADC code returns a value that is higher than expected by about 12%.  At least the value is consistent.)

closed 2021-08-17 by gigamegawatts

mr-pl3r, 2022-01-17:
Hi @gigamegawatts ,

How did you end up measuring the A0 reading? 

I have tried the next formula to get the data, but it's still not working for me...first I identify the g_vref with the next code and then use it on the formula to get the aprox voltage.  
`
float g_vref = 24637;  //Two Point --> coeff_a:24637mV coeff_b:0mV   <<<<< This is the result I always get

void adc_vref_init(){
    esp_adc_cal_characteristics_t adc_chars;
    esp_adc_cal_value_t val_type = esp_adc_cal_characterize((adc_unit_t)ADC_UNIT_1, (adc_atten_t)ADC1_CHANNEL_2, (adc_bits_width_t)ADC_WIDTH_BIT_12, 1100, &adc_chars);
    //Check type of calibration value used to characterize ADC
    if (val_type == ESP_ADC_CAL_VAL_EFUSE_VREF) {
        Serial.printf("eFuse Vref:%u mV\n", adc_chars.vref);
        g_vref = adc_chars.vref;
    } else if (val_type == ESP_ADC_CAL_VAL_EFUSE_TP) {
        Serial.printf("Two Point --> coeff_a:%umV coeff_b:%umV\n", adc_chars.coeff_a, adc_chars.coeff_b);       
    } else {
        Serial.println("Default Vref: 1100mV");
    }
}
//This formula is what I use to get the voltage - but it's not consistent and returning readings all over the place.
float v = analogRead(A0);
float battery_voltage = ((float)v / 4095.0) * 2.0 * 3.3 * (g_vref / 1000.0);
`

Thank you in advance for your help,

gigamegawatts, 2022-01-17:
Hi @mr-pl3r ,

I based my code on the ESP-IDF's example (in the ESP-IDF folder, it's in examples\peripherals\adc\esp32c3\adc).

It uses code that is different from the example for the ESP32 and ESP32-S2.  

My code is based on the esp32c3 example's single_read() function:
```

static int read_battery()
{
    int32_t adc_total = 0;

    adc1_config_width(ADC_WIDTH_BIT_DEFAULT);
    adc1_config_channel_atten(ADC1_CHANNEL_2, ADC_ATTEN_DB_11);

    int n = 20;
    while (n--) {
        adc_total += adc1_get_raw(ADC1_CHANNEL_2);
    }

    int average_adc = adc_total / 20;
    ESP_LOGI(TAG, "raw battery adc %d", average_adc);

    return average_adc;
}
```

The formula I used to convert the adc reading to volts is 6*average_adc/4095.  I think I just made up that formula, but it was close enough for my purposes.

Dan.

mr-pl3r, 2022-01-17:
@gigamegawatts  Thank you for the info, I will give it a shot with your code and report back if I see something wrong. 
Take it easy!

mr-pl3r, 2022-01-18:
@gigamegawatts 
After some testing I was able to come up with the error % , therefore the formula I have is :  

  float battery_voltage = ((2*3.3*v)/4095)-.23;

BTW - I needed to change this line 
From: 
adc1_config_width(ADC_WIDTH_BIT_DEFAULT); 
To:
adc1_config_width(ADC_WIDTH_12Bit); //since SC3 is always 12 bit and it was throwing an error on Arduino IDE according to the espressif document shown below.

https://github.com/espressif/esp-idf/blob/187e6ff66f653ac21fb6ab67e3cc8cc3a571b032/components/driver/adc.c#L668

gigamegawatts, 2022-01-20:
@mr-pl3r  I'm glad you got it to work.  

I didn't realize you were using Arduino.  I thought the Arduino code was taken from ESP-IDF, but I guess some things are different.   I'm using ESP-IDF with Espressif's build system, and ADC_WIDTH_BIT_DEFAULT is the same as ADC_WIDTH_BIT_12 (3).  You're right: only 12-bit is supported by the ESP32-C3. 

I originally used ESP-IDF 4.3.0.  Out of curiosity, I just tried the latest release, ESP-IDF 4.4 RC1.  In this release they've refactored the ADC code, so the ESP32-C3 example I originally used doesn't exist.  I used the closest thing, an example named single_read.

It still works OK, and adc1_get_raw() returns similar values to 4.3.0.  The value jumps around a lot (by about 3%), so it's still necessary to read it a bunch of times and average the result.  The single_read example calls esp_adc_cal_raw_to_voltage(), but the result isn't even close.

gigamegawatts, 2022-02-10:
@roomme13  I guess you're referring to my comment about esp_adc_cal_raw_to_voltage() not working properly?

Sorry, I was exaggerating when I wrote "isn't even close".  I just tested again, and that function returned an average of 2124.  My multimeter measured 2063, so that's not bad, about 3% too high.   (It returns half the actual battery voltage, since the T-IO-PLUS uses a 1:1 voltage divider to connect the battery to A0.  The actual battery voltage measured by my multimeter was 4.13V)

I don't know what accounts for the 3% error.  It might be a software problem in their SDK, but not necessarily.  It might be  related to the "eFuse" parameters that are burned into the chip.  My T-IO-PLUS is an older one, with the board dated 20210716.  LilyGo released a newer version in September -- I don't have it, but perhaps I would get a better result with that one.  

3% is probably close enough when measuring the battery.  For other purposes where you need a more accurate measurement, the easiest solution is to use a separate ADC breakout board with a chip like the MCP3002.