Xinyuan-LilyGO/LilyGo-T-OI-PLUS #6: Turn off power LED

dogsop, 2021-08-16:
Can the power LED be disabled in software to cut down the power drain when running on battery?

mgsb, 2021-08-16:
The schematic (https://raw.githubusercontent.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS/main/schematic/T-OI_PLUS_Schematic.pdf) shows the red LED hardwired to Vdd33.

dogsop, 2021-08-17:
> The schematic (https://raw.githubusercontent.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS/main/schematic/T-OI_PLUS_Schematic.pdf) shows the red LED hardwired to Vdd33.

The schematic is wrong, at least for the version I got. The red LED is wired to a GPIO pin, the green LED is wired to power. I was having trouble following the schematic and was hoping that it was also possible to turn off the green LED.

gigamegawatts, 2021-08-17:
I noticed that the photos of the board on Lilygo's AliExpress site are of an older version of the board.  Their photos show a board labelled 20210625, with the green LED in the middle.  

My board is labelled 20210716, version 1.1.  It has the green LED on the outside and the red LED in the middle.  On this board, the green LED is controlled through GPIO 3, while the red LED is always on when the board is powered.  So, the schematic seems to be for the newer version of the board.

Incidentally, LilyGo's AliExpress page for this board now says "Sorry, this item is no longer available!"  I don't know if that means out-of-stock, or discontinued.

mgsb, 2021-08-17:
Yes, I have v1.1 board, too, which appears to match the schematic. The LEDs, from left to right are:
* Green - controlled by GPIO03
* Red - power
* Blue - USB (charging)

al1fch, 2021-08-28:
OK with @gigamegawatts and @msg for V.1.1  board (2021 07 16)

Power red led is easy to remove...

LilyGO, 2021-09-30:
If you don't want to stick red power lights, you can note with customer service.

closed 2021-10-19 by LilyGO