Xinyuan-LilyGO/LilyGo-T-OI-PLUS #14: Access Grove Connector

vince925, 2021-12-02:
I am trying to read data from Grove connector (Multichannel Gas Sensor v2.0).  There is an LED that lights up when connected to a Seeed Device (and I can read data).  On my LILYGO mini D1 Plus v1.2 (ESP32C3), it is not lighting up nor do I see any data on SDA/SCL.  I measured voltage - v3.28 which is fine.  I am trying to read SCL (18) and SDA (19). Has anyone gotten the Grove connector to work?  Could I get an example?
Thanks in Advance, Vince

MadGeometer, 2022-02-20:
I'm also having problems with the Grove connector on T-OI Plus. I'm using a known working device with I2C address 0x3E. Also using SDA=19 and SCL=18.

I'm powering T-IO Plus using USB, but the I2C device doesn't seem to be getting any power. 

Any ideas?

vince925, 2022-02-20:
Mike,

     It sounds like you are running into the same issue as me.  I have a grove connector device that has a little led power indicator.  It lights up when connected to a WIO Terminal, but not when connected to the T-IO Plus.  I checked the power from the T-IO Plus and read 3.28v.  I did not check the AMPS which in hind sight I should have.  I connected another device to the main board on the T-IO (not the grove connector) and it worked.  I will check the AMPS on the grove connector on Tuesday and let you know.  I want to use the T-IO Plus as it has perfect form for several of my projects.  I was hoping a new firmware release would help, but have not seen any.  I tried contacting the folks at Lilygo and did get a response but it did not help.  If you could try contacting them that may help.  Let me know if you have any suggestions.

Vince

MadGeometer, 2022-02-20:
Hi Vince,

I'd interested in what you find out! I got a good deal on the T-OIs and am looking forward to using them!

Thanks,
Mike

LilyGO, 2022-02-22:
I use Grove to use everything fine.  Can you scan the I2C address now?  No electricity?  Is the sensor connected to 3.3V?

vince925, 2022-02-23:
I have tried scanning for I2C devices - if I scan without a device attached, it returns no devices found.  If I attach a device, it will hang or crash the T-IO Plus.  I have tried a couple mini D1 plus.  I have also tried several I2C devices.  I also uploaded the firmware "LilyGo-T-OI-plus_2021-7-17" with no success.  I am using seeedstudio devices (Multichannel Gas Sensor v2.0, HCHO Sensor v1.0, Temperature & Humitity Sensor Pro v1.3.  Have you used any seeedstudio grove devices?  I want T-IO's to work because they are perfect form and concept for my projects.  Please help, Vince

LilyGO, 2022-02-23:
#include "Wire.h"
#include "Arduino.h"

#define SDA 18
#define SCL 19

void setup()
{
    Serial.begin (115200); // Leonardo: wait for serial port to connect
    while (!Serial) { }
    Serial.println ();
    Serial.println ("I2C scanner. Scanning ...");
    unsigned char count = 0;
    Wire.begin(SDA, SCL);
    for (unsigned char i = 8; i < 120; i++) {
        Wire.beginTransmission (i);
        if (Wire.endTransmission () == 0) {
            Serial.print ("Found address: ");
            Serial.print (i, DEC);
            Serial.print (" (0x");
            Serial.print (i, HEX);
            Serial.println (")");
            count++;
            delay (1); // maybe unneeded?
        } // end of good response
    } // end of for loop
    Serial.println ("Done.");
    Serial.print ("Found ");
    Serial.print (count, DEC);
    Serial.println (" device(s).");
} // end of setup
void loop() {}

LilyGO, 2022-02-23:
What does the serial port print using the above code?

vince925, 2022-02-26:
I received the following output (COM14 / USB-C):

 

09:36:18.793 -> ESP-ROM:esp32c3-api1-20210207

09:36:18.793 -> Build:Feb  7 2021

09:36:18.793 -> rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)

09:36:18.793 -> SPIWP:0xee

09:36:18.793 -> mode:DIO, clock div:1

09:36:18.793 -> load:0x3fcd6100,len:0x420

09:36:18.793 -> load:0x403ce000,len:0x90c

09:36:18.793 -> load:0x403d0000,len:0x236c

09:36:18.793 -> SHA-256 comparison failed:

09:36:18.793 -> Calculated: ccb0d00bac7e84e1d90a12e4f75f4ab6c5f7e71bb209afd5819c4c9557a6db71

09:36:18.839 -> Expected: c9cf160580940ec7801c73b16423824e72ad12055c732e83ce66332240af42a7

09:36:18.839 -> Attempting to boot anyway...

09:36:18.839 -> entry 0x403ce000

09:36:18.886 -> 

09:36:18.886 -> I2C scanner. Scanning ...

09:36:24.881 -> E (11092) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

09:36:24.881 -> E (11092) task_wdt:  - IDLE (CPU 0)

09:36:24.881 -> E (11092) task_wdt: Tasks currently running:

09:36:24.881 -> E (11092) task_wdt: CPU 0: loopTask

09:36:29.894 -> E (16092) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

09:36:29.894 -> E (16092) task_wdt:  - IDLE (CPU 0)

09:36:29.894 -> E (16092) task_wdt: Tasks currently running:

09:36:29.894 -> E (16092) task_wdt: CPU 0: loopTask

09:36:34.892 -> E (21092) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

09:36:34.892 -> E (21092) task_wdt:  - IDLE (CPU 0)

09:36:34.892 -> E (21092) task_wdt: Tasks currently running:

09:36:34.892 -> E (21092) task_wdt: CPU 0: loopTask

09:36:39.850 -> E (26092) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

09:36:39.898 -> E (26092) task_wdt:  - IDLE (CPU 0)

 

It shows firmware 2/7/2021 Temperature & Humidity Sensor Pro v1.3 was attached to the T-IO Plus.  I was using Arduino v1.8.19.  I cut/paste your code into the gui.

MadGeometer, 2022-02-26:
Hi Vince and LilyGO

OK, got it figured out...
The pins on the T-OI board's Grove connector are backwards from the usual Grove connector!

Order on T-OI is GRD, 3V3, SCL, SDA
Order on usual Grove sensor is SCL, SDA, VCC, GND

Attached photo for proof. 

Thanks,
Mike K

[image]

vince925, 2022-02-26:
Mike,

     It is totally assume that you were able to figure out the wiring / connection issue.  LilyGo will have to correct this probably with a hardware change.  In the meantime, how would you suggest to correct for the issue – make a short cross over cable or just re-pin the cable?

Great detective work, Vince

 

 <https://user-images.githubusercontent.com/3835813/155821449-001d43a8-4f5e-4b9a-aef3-84ce66d16206.jpg>

vince925, 2022-02-26:
> Hi Vince,
>
> Just tried the following trick - reverse the values of SDA and SCL in the constructor for Wire or the particular device. I'm using the Color LCD display from M5Stack, which is an I2C device that would ordinarily be initialized like:
>
> #define I2C_SDA 19
> #define I2C_SCL 18
> M5UnitLCD display = M5UnitLCD(I2C_SDA, I2C_SCL);
>
> Instead, do this:
>
> #define I2C_SDA 19
> #define I2C_SCL 18
> M5UnitLCD display = M5UnitLCD(I2C_SCL, I2C_SDA);
>
> The numbers 19 and 18 can be swapped, too.
>
> Since this works, the cable's end connectors can be filed-down and then reversed, no cross-over required. I don't know what the downstream effects of switching SCL and SDA ports are, though.

Mike,

     From your picture, more is swapped.  It is almost a mirror image?

 

From T-IO Plus side:

    Black    -> White,

    Red      -> Yellow

    White  -> Red

    Yellow -> Black

 

Vince

MadGeometer, 2022-02-26:
Yes, it is NOT a reversal

mailinger-mate, 2022-10-25:
@LilyGO any chance a newer version of the board will receive a fix for the Grove wiring?

LilyGO, 2022-11-14:
Do you mean we have a mistake in this image recognition? We've finished changing it
[image]

vince925, 2022-11-14:
LilyGO,
     I saw that there is a new version of LilyGo-T-OI-PLUS  (2022-06-20).  What is different.  Even changing the pinout on a V1.2 board (20210922 with firmware update), I have not been able to get a Grove device to work (I have 2: Multichannel Gas Sensor v2.0 and Temperature & Humidity Sensor Pro v1.3).  The sensors work on a WIO Terminal.  Neither work using Seeed Library / demo

LilyGO, 2022-11-14:
Just change the serial chip. You make sure your code SDA and SCL correspond to your wiring.

mailinger-mate, 2022-11-19:
@LilyGO according to the [latest pinout on master](https://github.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS/blob/main/image/new_TOI_Plus.png) the Grove mount wiring is still *incorrect*.

The [Grove interface specifications](https://wiki.seeedstudio.com/Grove_System/#interface-of-grove-modules) defines the correct wiring as:
- pin 1 - Yellow (for example, SCL on I2C Grove Connectors)
- pin 2 - White (for example, SDA on I2C Grove Connectors)
- pin 3 - Red - VCC on all Grove Connectors
- pin 4 - Black - GND on all Grove Connectors

However as @vince925 also pointed out on the T-OI board the wiring is **incorrectly the mirrored opposite**, and since the connector is polarized, and the currently the yellow SCL/SDA signal wire can only ever be inserted to GND on the T-OI.

For example, on the [SW-420 Grove vibration sensor pin map](https://wiki.seeedstudio.com/Grove-Vibration_Sensor_SW-420/#pin-map) you can see can see how the Pin 1 wire is supposed to be SIG for vibration signal:

[image]

And to illustrate the vibration sensor connected to a T-OI with the pin map displayed, maybe the problem seems more obvious:

[image]

Is this something I can also fix in code?

vince925, 2022-11-24:
[image]

I am using the multi gas sensor from Seeed Studio.  It is finally working.  The T OI Plus would short out if I had the power backwards.  WireScan would not find the sensor until I wired the cable as seen in the picture (it seemed to work when I used individual wires - but red and black were reversed which makes not sense).  Now that it works - this will be the best solution (better than the Wemos C3 mini) - it just should not be this hard.

vince925, 2022-11-24:
No - since power and SDA/SCL are reversed it can not be corrected in software alone

vince925, 2022-11-24:
[image]

I am using the multi gas sensor from Seeed Studio. It is finally working. The T OI Plus would short out if I had the power backwards. WireScan would not find the sensor until I wired the cable as seen in the picture (it seemed to work when I used individual wires - but red and black were reversed which makes no sense). Now that it works - this will be the best solution (better than the Wemos C3 mini) - it just should not be this hard.

vince925, 2022-11-25:
[image]

I tried a DHT sensor from Seeed Studio.  I had to swap the yellow & white pins on the T OI Plus side (power / ground)  This does not make sense to me - but it worked.

mailinger-mate, 2022-12-03:
> [image]
>
> I am using the multi gas sensor from Seeed Studio. It is finally working. The T OI Plus would short out if I had the power backwards. WireScan would not find the sensor until I wired the [...]

@vince925 nice job, i am just wondering, where did you get the Groove connector with the reverse wiring you shared the first photo of? Did you make it yourself?

vince673, 2022-12-03:
I removed the pins on the cable side used to plug into the T OI Plus (Grove).  Then I plugged them back in the order needed.

vince673, 2022-12-03:
I forgot to mention - there is something screwy going on - You may need a different configuration, but overall this is a nice board with good connections - just needs more work by LilyGO designers

mailinger-mate, 2022-12-03:
> I removed the pins on the cable side used to plug into the T OI Plus (Grove). Then I plugged them back in the order needed.

Hehe, great idea!

JpEncausse, 2024-03-21:
Question to Grove / P-OI experts : 
Is it possible to go deepsleep on battery then wakeup from grove sensor ?
[I'd love to do something like this guy](https://www.youtube.com/watch?v=he_Z-mDLnlM) with a Contact / PIR / ToF sensor to detect a door open.

mailinger-mate, 2024-06-16:
@JpEncausse have you managed to trigger wakeup from grove sensor?

Ivoz, 2025-01-19:
Yes, just found this out. The board's Grove connector connected the SDA and SCL in the wrong order. You'll have to re-wire your cable to work with this board