Xinyuan-LilyGO/LilyGo-T-OI-PLUS #18: Can not program the board when Li-Po battery is attached

pymo, 2022-04-05:
When Li-Po battery is attached to the board, and the board connects to PC via USB, the PC (using Arduino) can not find the COM port, thus not able to program the board. After desoldering the battery, the COM port appears and program succeeded.

This is annoying because I had to desolder the battery everytime I change the program.

LilyGO, 2022-04-06:
Hello, I have tried, there is no problem you said.  How do you connect the battery?  Can you take a picture for me?

pymo, 2022-04-20:
Sorry, it turns out to be my USB hub's problem. It does not provide enough current. I plugged it to the USB port directly on the motherboard and it works fine.

closed 2022-04-20 by pymo