Xinyuan-LilyGO/LilyGo-T-OI-PLUS #19: Measuring Vbat

ckuehnel, 2022-05-26:
I want to measure the voltage of Vbat and use the folllowing program:

```
void setup() 
{
  Serial.begin(115200);
  
  analogReadResolution(12);
  analogSetAttenuation(ADC_11db); 
}

void loop() 
{
  // read the analog / millivolts value for GPIO2:
  int analogValue = analogRead(2);
  int analogVolts = analogReadMilliVolts(2);
  float temp = temperatureRead();
  
  // print out the values you read:
  Serial.printf("ADC analog value = %d\n",analogValue);
  Serial.printf("ADC millivolts value = %d\n",analogVolts);
  Serial.printf("Core temperature = %3.1f °C\n\n", temp);
  
  delay(2000);  // delay in between reads for clear read from serial
}
```
The output is:
E (798844) tsens: Do not configure the temp sensor when it's running!
E (798844) tsens: Temperature sensor is already running or not be configured
ADC analog value = 2954
ADC millivolts value = 1407
Core temperature = 44.6 °C

I expect a voltage of about 2000 mV for a full loaded battery.What means the tsens output?

am009, 2022-05-28:
I am also trying to measure the voltage of Vbat. According to my previous experience on T-Display board, the maximum measuring range of the esp32 series chip is 0 mV ~ 2500 mV (https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32c3/api-reference/peripherals/adc.html ). So, for a 3.7V battery, usually the adc pin only measures half of the voltage:
[image]
By looking at the schematic, this seems to be the case. the connection is GND -> R13(100K) -> R7(0) -> A0 -> R26 -> Vbat, and the adc is mesuring the volt between R13 and GND. So, simply double the voltage.

am009, 2022-05-28:
My current code to get volt is:

```c

// see https://github.com/espressif/esp-idf/blob/master/examples/peripherals/adc/single_read/single_read/main/single_read.c
esp_adc_cal_characteristics_t adc_chars;

void battery_adc_init()
{
  esp_err_t ret;
  bool cali_enable = false;
  esp_adc_cal_value_t val_type;
  ret = esp_adc_cal_check_efuse(ESP_ADC_CAL_VAL_EFUSE_TP);
  if (ret == ESP_ERR_NOT_SUPPORTED) {
      Serial.println("Calibration scheme not supported, skip software calibration");
  } else if (ret == ESP_ERR_INVALID_VERSION) {
      Serial.println("eFuse not burnt, skip software calibration");
  } else if (ret == ESP_OK) {
      cali_enable = true;
  } else {
      Serial.println("Invalid arg");
  }

  val_type = esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 0, &adc_chars);
  if (val_type == ESP_ADC_CAL_VAL_EFUSE_TP)
  {
    Serial.printf("Two Point --> coeff_a:%umV coeff_b:%umV\n", adc_chars.coeff_a, adc_chars.coeff_b);// Two Point --> coeff_a:47304mV coeff_b:0mV
  }
  else
  {
    Serial.println("Unknown Calibration");
  }

  adc1_config_width(ADC_WIDTH_BIT_12);
  adc1_config_channel_atten(ADC1_CHANNEL_2, ADC_ATTEN_DB_11); // ADC1_CHANNEL_2 => GPIO2
}

uint32_t get_battery_mvolt() {
  int raw = adc1_get_raw(ADC1_CHANNEL_2); // GPIO2
  uint32_t volt = esp_adc_cal_raw_to_voltage(raw, &adc_chars); // mVolt
  return volt;
}
```

There is no `#ifdef CONFIG_IDF_TARGET_ESP32C3` in https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-adc.c . I guess directly using analogReadMilliVolts will return an uncalibrated result.

ckuehnel, 2022-05-28:
Thats clear. I should meassure the voltage output via serial monitor at A0. But, the values differ. Dr. Claus Kühnel - Talstr. 13b - CH-8852 AltendorfMobile +41-76-3960801

ckuehnel, 2022-05-28:
/*
 * File: ESP32-C3_AnalogRead.ino
 * 
 * Using ADC of T-OI plus ESP32-C3
 * used Board: STAMP-C3U
 * 
 * 2022-05-27/Claus Kühnel ***@***.***
 */
 
float conversionFactor = 3300./4096;

void setup() 
{
  Serial.begin(115200);
  delay(2000); // wait for serial monitor
  Serial.println("T-OI plus ESP32-C3 ADC");
  
  analogReadResolution(12);
  analogSetAttenuation(ADC_11db); 
}

void loop() 
{
  // read the analog value for GPIO2:
  int analogValue = analogRead(2); // Reads Vbat
  // int analogVolts = analogReadMilliVolts(2); do not use this 
  int analogVolts = (int) (analogValue * conversionFactor) + .5;
  float temp = temperatureRead();
  
  // print out the values you read:
  Serial.printf("ADC analog value = %d\n",analogValue);
  Serial.printf("ADC millivolts value = %d\n",analogVolts);
  Serial.printf("Core temperature = %3.1f °C\n\n", temp);
  
  delay(2000);  // delay in between reads for clear read from serial
}

I get the following output for 2.06 V on A0:

Dr. Claus Kühnel - Talstr. 13b - CH-8852 Altendorf
Mobile +41-76-3960801
-- Aus dem Rheinwerk-Verlag: Arduino - das umfassende Handbuch ISBN 978-3-8362-7345-9 [https://www.rheinwerk-verlag.de/arduino-das-umfassende-handbuch/] -- 

[image] [https://user-images.githubusercontent.com/23314210/170810678-96a8d7c7-91a8-47b8-952e-5ebf32fca1eb.png]
[cf2c34dd-2ffc-4736-b131-183d6f40f7ac]

LilyGO, 2022-05-30:
Hello,
1. [Obtain battery voltage](https://github.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS/tree/main/example/battery_voltage).
2.The output is:
E (798844) tsens: Do not configure the temp sensor when it's running!
E (798844) tsens: Temperature sensor is already running or not be configured

This may have something to do with the ESP32 version. Or a software bug
 Submit feedback questions [here](https://github.com/espressif/arduino-esp32/issues)

ckuehnel, 2022-05-30:
Thanks a lot, it works now.

Dr. Claus Kühnel - Talstr. 13b - CH-8852 Altendorf
Mobile +41-76-3960801
-- Aus dem Rheinwerk-Verlag: Arduino - das umfassende Handbuch ISBN 978-3-8362-7345-9 [https://www.rheinwerk-verlag.de/arduino-das-umfassende-handbuch/] -- 

Hello,

[40b7a09a-1642-4976-9865-96118185cffb]

closed 2022-05-31 by LilyGO

factoidforrest, 2023-05-10:
Soo Is vbat hardwired to one of the ADCs, or did you loop a wire from vbatt back to a pin?