Xinyuan-LilyGO/LilyGo-T-OI-PLUS #20: Deep sleep wake up by touch

mailinger-mate, 2022-06-25:
Hello!

I would like to be able to wake up the board from deep sleep using touch following this tutorial:
https://randomnerdtutorials.com/esp32-touch-wake-up-deep-sleep/

It seems the `esp_sleep_enable_touchpad_wakeup()` method is not available using PlatformIO and the Arduino framework.

Is it possible to achieve some kind of a wake up using the following method?
`  esp_deep_sleep_enable_gpio_wakeup(1ULL << 3,ESP_GPIO_WAKEUP_GPIO_HIGH);`

Can you please help me with writing a code to wake up the board with touch or an external vibration sensor?
https://www.electroniclinic.com/shock-sensor-arduino-circuit-code-ky-002-shock-sensor/

Thank you!

liangyingy, 2022-06-25:
No touch sensors on ESP32-C3

liangyingy, 2022-06-25:
```c
#include <Arduino.h>
#include "esp_sleep.h"

void setup()
{
    Serial.begin(115200);
    delay(100);

    const int wakeup_time_sec = 20;
    printf("Enabling timer wakeup, %ds\n", wakeup_time_sec);
    esp_sleep_enable_timer_wakeup(wakeup_time_sec * 1000000);

    const int ext_wakeup_pin_0 = 2;
    printf("Enabling EXT0 wakeup on pin GPIO%d\n", ext_wakeup_pin_0);
    esp_deep_sleep_enable_gpio_wakeup(1 << ext_wakeup_pin_0, ESP_GPIO_WAKEUP_GPIO_LOW);
    // esp_sleep_enable_ext0_wakeup(ext_wakeup_pin_0, 1);

    esp_deep_sleep_start();
}

void loop()
{
    delay(100);
}
```

mailinger-mate, 2022-06-26:
Thank you, meanwhile i also found the Espressif documentation on the differences:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/chip-series-comparison.html

mailinger-mate, 2022-06-26:
Does  your code  example wake up the board from deep sleep when the GPIO pin #1 turns to low?
Is that pin actually exposed on the board according to the pinout?
https://github.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS/blob/main/image/new_TOI_Plus.png

liangyingy, 2022-06-27:
The code is fine, the code uses gpio2.

mailinger-mate, 2022-07-09:
Thank you, the code you suggested code is working!

closed 2022-07-09 by mailinger-mate