Xinyuan-LilyGO/LilyGo-T-OI-PLUS #21: USB & Battery high power consumption during deep sleep

mailinger-mate, 2022-07-09:
Hello,

Based on #8 I confirmed the power consumption on *battery only* to be *40 uA* during deep sleep.

However with a fully charged *battery and USB* both connected the power consumption never goes below *35 mA*.
I am using the [deep sleep example](https://github.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS/blob/main/example/deepsleep/deepsleep.ino) provided in the repository.

How can i have external power with backup battery and charging while preserving low power consumption in deep sleep?

Thank you,
Mate

LilyGO, 2022-07-19:
Hello, Test power can only use battery test !

closed 2022-07-19 by LilyGO

mailinger-mate, 2022-07-23:
@LilyGO what alternatives do I have?

ekawahyu, 2023-03-08:
@mailinger-mate I am sorry for seeing your message late. I am afraid you are gonna need a new design on the power management part. I hope that helps!

mailinger-mate, 2023-03-19:
@ekawahyu will Lilygo release a board with an improved or alternative design that fixes this?

ekawahyu, 2023-03-19:
I think that question is for LilyGo to answer. In my case, I made my own design to fit my requirement, with better low quiescent LDO and for some I completely remove it.