Xinyuan-LilyGO/LilyGo-T-OI-PLUS #25: How to use external antenna? Will external antenna boost strength of sending signal?

raoulshih, 2022-09-02:
Hi All

I am trying using this device as a WiFi-tag, however the strength of the signal is a bit weak given its small antenna. 
How should I do to enable use of external antenna?  For other development boards it can be done by de-soldering and re-soldering one resistor to another location, however for this board I was unable to find the information. Please advice and thanks in advance.

LilyGO, 2022-11-14:
Hello，
The same method of use. Switch the resistor to use external antenna as shown
[image]

closed 2022-11-14 by LilyGO

Goldthing, 2026-08-19:
@LilyGO could this please be included in the main documentation. Despite looking through the repo I had to find this information by searching the internet only to be returned to it here. Thanks.