Xinyuan-LilyGO/LilyGo-T-OI-PLUS #29: Expected battery life

wolfen351, 2023-01-04:
I am doing some measurements on my board, and thought I would share my findings about battery life:

1) Running a micropython program doing lots of calculations and about 50 temperature readings a second on a DS18B20 temperature sensor, Red LED intact, Green LED on 50% of the time, alternating on and off each cycle of the main program, with Wifi connected. No sleep code, CPU at full speed.

Battery life: 1h 20min

2) Running a micropython program doing lots of calculations and about 10 temperature readings a minute on a DS18B20 temperature sensor, Red LED intact, Green LED on 50% of the time, alternating on and off each cycle of the main program, with Wifi connected. Awake for 1 second, LIGHT sleep for 60 seconds. CPU at full speed.

Battery life: 1h 25min

I'm working on getting to 5h of battery life, with RAM intact. Next up, much longer sleep 10 min with 1 second of work each time. I'll also remove the RED led, as by all accounts that seems to eat power! Also planning on disabling the greed LED to save power.

wolfen351, 2023-01-08:
Battery life: 3h+ (still running, battery reports as 95%)

Still have the red led on. I've disabled the green LED while sleeping, and flash it at 50% when working. I only reed the DS18B20 every 5 min, and sleep for 4 min at a time. Stay awake for about 1.5s to reconnect to the network and send MQTT messaqes. The big power saving from 1h25min to now appears to be because of reading the temp sensor much less often. Still doing light sleep to preserve ram contents.

wolfen351, 2023-01-15:
Battery life: 5d

New update, I did a lot to optimise my code and reduce the work done when awake. Also made sure to reconnect to wifi/mqtt if they dropped while sleeping. I think this caused me to think the board had run out of power when in fact it had just lost wifi and never reconnected.

Terrence77, 2023-07-04:
@wolfen351 6days, that is great.  Would you mind sharing your code?