Xinyuan-LilyGO/LilyGo-T-OI-PLUS #33: Does not run on Battery

Terrence77, 2023-07-04:
I have a simple blink script.

runs fine while plugged into usb, but does not run on battery.

Is there some setting I am missing to enable the PLUS to run on battery?

Thank you.

Terrence77, 2023-07-06:
Hello, is anyone from LilyGo monitoring this Issues area?

lozaning, 2023-09-29:
The + terminal from the battery holder connects to a through hole labeled 'G' on the PCB. Did you end up swapping the polarity of the battery? Mine wont run from batteries either.

aalexff, 2024-10-31:
I'm powering it with a power supply at 4.2V connected directly on the battery terminals and it works fine.
But I haven't been able to make it work with an actual battery, it seems it's really unstable below the 4.2V mark, as 3.8V to 4.1V may or may not work.
My guess is that it has a very narrow window of working voltage,